makerbase-mks/MKS-Robin #205: Serial port issue with Robin Lite

jaymcd, 2021-09-13:
I am getting errors with serial ports on compile --

 In file included from Marlin\src\HAL\STM32\MarlinSPI.h:24,
                 from Marlin\src\HAL\STM32\MarlinSPI.cpp:26:
Marlin\src\HAL\STM32\HAL.h:65:4: error: #error "SERIAL_PORT must be from 1 to 6. You can also use -1 if the board supports Native USB."
   65 |   #error "SERIAL_PORT must be from 1 to 6. You can also use -1 if the board supports Native USB."

Not sure if this is a vscode issue or a platform issue or a board/library issue. It was obviously working before as the default config seems to be for -1 or 0. I have tried both with no success. It will compile if I change to value 1 -6 ....

fjbold1, 2022-06-19:
I was able to use -1, not at first but after enabling native usb in the adv file.

k2ksk2k, 2023-04-28:
According to MKS-Robin, you can try the settings "#define SERIAL_PORT 3" and "#define SERIAL_PORT_2 1".